Embroidermodder/libembroidery #1: System/OS

edgarsstrods, 2017-02-16:
Is this library made for specific OS/purpose and meant to be public? I am not able to 'make' this under Ubuntu 14.04. Seems to have some hardcoded paths to disk labeled "E:\Qt\...", Windows style.

Are there any major changes if compared to the libembroidery-convert under https://github.com/Embroidermodder/Embroidermodder ?

Thank you!

JoshVarga, 2017-02-16:
This is intended to be public, all that I have really done is separate this
from Embroidermodder. It was originally intended to be used with "qmake"
and then "make"

I intend to create a proper makefile and auto configurations etc. I haven't
gotten around to this yet.

edgarsstrods, 2017-02-17:
Ok, now I see - first "qmake". Thank you!
If this is just a seperation, then the library is the same as in "big project", no updates I guess.

closed 2021-08-06 by JoshVarga