Embroidermodder/libembroidery #2: Does not recognize DST trim command.

tatarize, 2017-10-27:
DSTs have a hacky sort of trim. They do series of jumps. 
Jump: +2 +2
Jump: -4 -4
Jump: +2 +2

The net result is a jump of zero but this triggers a machine's trim. 

Without libembroidery accepting that or allowing all jumps to be imply a trim (allow an option maybe) for DST it just makes stitches across the converted embroidery.

tatarize, 2017-11-10:
Turns out it's a bit simpler than that. Machines automatically trim after a number of jumps. If you add three jumps it tends to trim anyway. So basically 1 or 2 jumps will sew but between 3-8 or so it will always trim. So one could reasonably just imply a trim for any series of jumps because that's what the machines will do with that.

JoshVarga, 2017-11-10:
Right now in libembroidery this is commented out since it wasn't well tested. If you look in format-dst.c there are the following lines... (which assume 5 jumps is enough, but I believe is configurable on many machines)

```
/* if(flags & TRIM)
    {
        int v = 5;
        int dx = (int)(x/v), dy = (int)(y/v);
        for(i = 1; i < v; i++)
        {
            encode_record(file, dx, dy, JUMP);
        }
        encode_record(file, x - (dx * (v - 1)), y - (dy * (v - 1)), JUMP);
        return;
    } */
```

robin-swift, 2021-11-07:
I'm making an executive decision to just include this code.

closed 2021-11-07 by robin-swift