Embroidermodder/libembroidery #9: Download of libembroidery-convert

robin-swift, 2021-07-28:
I think when this project has reached a certain level of completion (in terms of file type support) it would be good to have a binary download for anyone that wants to convert between embroidery file types but doesn't want to use the library in code.

Could we set up some milestones or something so I can know what would be needed for that as a contributor?

JoshVarga, 2021-07-28:
Embroidermodder has something similar to what I think you are describing...

https://github.com/Embroidermodder/Embroidermodder/tree/master/libembroidery-convert

robin-swift, 2021-07-28:
That's the same code (which means it's not clear which is the one being maintained as the "official one") as the **[libembroidery-convert-main.c](libembroidery-convert/libembroidery-convert-main.c)** in this directory. Given that it's a command line tool I think that it makes more sense that it is maintained here.

[The Embroidermodder website](http://embroidermodder.org/downloads.html) doesn't cover that for all platforms in either case, if it would be more useful I could help with getting version 2 shipping as an executable? That would help my stitchwork more than my use of this library directly but I'd need a good idea of what the roadmap is for the software (perhaps using the projects tab on that page).

robin-swift, 2021-08-04:
For example:

Image Magick uses github to make the [binary releases](https://github.com/ImageMagick/ImageMagick/releases). Backending to [runs like this one](https://github.com/ImageMagick/ImageMagick/actions/runs/1040108043).

Then those files could be put on the Embroidermodder website. We'd just need to write a file like [this one](https://raw.githubusercontent.com/ImageMagick/ImageMagick/main/.github/workflows/release.yml).

Here's some relevant lines from that example:
```
on:
  workflow_dispatch:
  push:
    branches:
    - main
    paths:
    - 'ChangeLog'

name: release
jobs:
  release_windows_installer:
    name: 'Windows ${{matrix.quantum}}${{matrix.hdri_flag}}-${{matrix.platform}} (${{matrix.type}})'
    runs-on: windows-latest

    strategy:
      fail-fast: false
      matrix:
        platform: [ x64, x86 ]
        type: [ dll, static ]
        quantum: [ Q8, Q16 ]
        hdri: [ HDRI, noHDRI ]
        exclude:
          - quantum: Q8
            hdri: HDRI
        include:
          - platform: x64
            bit: 64
            platform_alias: x64
          - platform: x86
            bit: 32
            platform_alias: Win32
          - type: dll
            type_flag: dmt
          - type: static
            type_flag: smtd
          - hdri: HDRI
            hdri_flag: '-HDRI'

    steps:
    - uses: actions/checkout@v2
      with:
        repository: ImageMagick/ImageMagick-Windows
        path: ImageMagick-Windows
        ref: refs/heads/main
        fetch-depth: 1

    - name: Clone repositories
      shell: cmd
      run: |
        cd %GITHUB_WORKSPACE%\ImageMagick-Windows
        CloneRepositories.cmd https://github.com/ImageMagick shallow

    - name: Configure ImageMagick
      shell: cmd
      run: |
        cd %GITHUB_WORKSPACE%\ImageMagick-Windows\VisualMagick\configure
        configure.exe /noWizard /VS2019 /includeOptional /installedSupport /deprecated /${{matrix.hdri}} /${{matrix.quantum}} /${{matrix.platform}} /${{matrix.type_flag}}

    - name: Build ImageMagick
      shell: cmd
      run: |
        call "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\Tools\VsDevCmd.bat"
        cd %GITHUB_WORKSPACE%\ImageMagick-Windows\VisualMagick
        set /p solution=<solution
        msbuild %solution% /m /t:Rebuild /p:Configuration=Release,Platform=${{matrix.platform_alias}}
  
    - uses: actions/upload-artifact@v2
      with:
        name: installers
        path: 'ImageMagick-Windows\VisualMagick\installer\output'
```
I hope this helps.

robin-swift, 2021-08-05:
I'll re-approach this with a PR, closing for now.

closed 2021-08-05 by robin-swift