Embroidermodder/libembroidery #17: Missing file: valgrind-create-suppressions.sh

robin-swift, 2021-07-31:
Hi

In the folder libembroidery-convert/ there's a script for running valgrind that refers to another script valgrind-create-suppressions.sh that isn't present. It looks to be a sed filter to tidy valgrind output.

Can you find this missing script, please?

There's one or two other "personal not public code" issues like `/* kate: bom off; indent-mode cstyle; indent-width 4; replace-trailing-space-save on; */` text editor configuration which I imagine can go somewhere else on the user's system (like a .bashrc equivalent?). For example [EditorConfig](https://editorconfig.org/) is a system that doesn't assume what your editor is. 

Cheers,

Robin

JoshVarga, 2021-08-01:
This code was moved from Embroidermodder so I believe it is referring to...

https://github.com/Embroidermodder/Embroidermodder/tree/master/valgrind-supp

Removing the text editor configuration would be good

robin-swift, 2021-08-01:
If both of these are accepted we can close this.

closed 2021-08-02 by JoshVarga