Embroidermodder/libembroidery #50: experimental directory

robin-swift, 2021-08-16:
The experimental code that has been copied over is still present in the Embroidermodder repo. Can that be removed there?

Do we need an entire copy if the guide code still? Can that be removed?

Also now we have CI in the form of Github Actions, can we remove all the Travis repositories? They confuse the project without adding new features.

So we'd have 5 repos: EmbroiderModder, libembroidery, the website, the help files and the robotics one.

JoshVarga, 2021-08-16:
It looks like the python experimental GUI code should be moved to a new repo or kept in Embroidermodder.

robin-swift, 2021-08-17:
Ok, those changes have dealt with most of that, are we keeping the Travis repositories?

JoshVarga, 2021-08-17:
We will get rid of the Travis repos, but I think we should get the actions working with the artifacts saved first

closed 2021-08-19 by robin-swift