Embroidermodder/libembroidery #69: How to generate the ddl to be used in windows forms c#

fabricocouto, 2021-09-08:
been trying to use the dll in c#
using swig to generate classes
but when you put the swig wrap.c file in the libembroidery dll project it simply gives an error of statements 
[image]

robin-swift, 2021-09-11:
Hi,

the bindings to C# aren't actively maintained so this likely broke with my current development of the library. Could you give me an idea of what you need them for and I can put it in the TODO section of the README.md?

Cheers,

Robin

fabricocouto, 2021-09-11:
I've been following Josh's work for many years, he even helps me a lot even though I already have all the native C# code to read and write the embroidery files but one I couldn't do in C# which is the .Hus I wanted, I can add it to the little project that I have it in C# but it is very complicated to write it in C# so I wanted to be able to use the dll if you want to char me directly here by email fabriciocoutodivino@gmail.com.
Josh already sent me the 64-bit dll but it doesn't open in my o32-bit project.
I would like to be able to create the dll myself because there are some errors in the pcs, pcm and xxx files that I can only solve by testing the code, which has already been corrected for reading in c# but in c only testing the dll
follow the image of my project to view and convert embroidery files 
[image]

robin-swift, 2021-09-11:
> "I would like to be able to create the dll myself"

Ah, in that case you could install CMake [from here](https://cmake.org/), then follow the build instructions in the README.md.

If you encounter a build issue then you can report that here as a new issue. Also you can report issues with converting from hus to pcs etc. here and they may get fixed and you can just use the `embroider` command line program that builds from this source.

fabricocouto, 2021-09-11:
by qt and using swig so don't compile only by cmake
the alone there said he just wanted to generate the dll in qt, because josh has been helping me when he has time there.
In fact, I would like to understand why the error happens to generate it here

robin-swift, 2021-09-11:
The `libembroidery.dll` file can be generated directly from this repository without the rest of the code in EmbroiderModder. That's partly why it was separated like this. That means that you can get the dll that way. As for SWIG I'm not actively maintaining it so you may have to submit an issue for how that has broken, as opposed to not building the dll and wait until that's the top priority.

fabricocouto, 2021-09-11:
this is the error that is generating in swing 
[image]

robin-swift, 2021-09-11:
I'm confused, there's nothing in the current source called `libembroidery-convert`, are you running the current version?

fabricocouto, 2021-09-11:
yes only the mone is modified there but it's the simplified font you uploaded

fabricocouto, 2021-09-11:
the error happens when i put this swig generated file in the dll project 
[swig_wrapt.zip](https://github.com/Embroidermodder/libembroidery/files/7148525/swig_wrapt.zip)

fabricocouto, 2021-09-11:
these are the lines with errors, when they are stripped it generates the dll but when it is called in the creat pattern it closes the program 

// error add in dll

SWIGEXPORT int SWIGSTDCALL CSharp_embArray_addSpline(void * jarg1, void * jarg2) {
  int jresult ;
  EmbArray *arg1 = (EmbArray *) 0 ;
  EmbSplineObject *arg2 = (EmbSplineObject *) 0 ;
  int result;
  
  arg1 = (EmbArray *)jarg1; 
  arg2 = (EmbSplineObject *)jarg2; 
  result = (int)embArray_addSpline(arg1,arg2);
  jresult = result; 
  return jresult;
}

// erro add in dll
SWIGEXPORT void * SWIGSTDCALL CSharp_embPathObject_create(void * jarg1, void * jarg2, void * jarg3, int jarg4) {
  void * jresult ;
  EmbArray *arg1 = (EmbArray *) 0 ;
  EmbArray *arg2 = (EmbArray *) 0 ;
  EmbColor arg3 ;
  int arg4 ;
  EmbColor *argp3 ;
  EmbPathObject *result = 0 ;
  
  arg1 = (EmbArray *)jarg1; 
  arg2 = (EmbArray *)jarg2; 
  argp3 = (EmbColor *)jarg3; 
  if (!argp3) {
    SWIG_CSharpSetPendingExceptionArgument(SWIG_CSharpArgumentNullException, "Attempt to dereference null EmbColor", 0);
    return 0;
  }
  arg3 = *argp3; 
  arg4 = (int)jarg4; 
  result = (EmbPathObject *)embPathObject_create(arg1,arg2,arg3,arg4);
  jresult = (void *)result; 
  return jresult;
}

// erro ind add dll
SWIGEXPORT void SWIGSTDCALL CSharp_embPathObject_free(void * jarg1) {
  EmbPathObject *arg1 = (EmbPathObject *) 0 ;
  
  arg1 = (EmbPathObject *)jarg1; 
  embPathObject_free(arg1);
}

// erro ind add dll
SWIGEXPORT void SWIGSTDCALL CSharp_embPathObject_free(void * jarg1) {
  EmbPathObject *arg1 = (EmbPathObject *) 0 ;
  
  arg1 = (EmbPathObject *)jarg1; 
  embPathObject_free(arg1);
}

// erro in add dll
SWIGEXPORT void * SWIGSTDCALL CSharp_embArcObject_create(double jarg1, double jarg2, double jarg3, double jarg4, double jarg5, double jarg6) {
  void * jresult ;
  double arg1 ;
  double arg2 ;
  double arg3 ;
  double arg4 ;
  double arg5 ;
  double arg6 ;
  EmbArcObject *result = 0 ;
  
  arg1 = (double)jarg1; 
  arg2 = (double)jarg2; 
  arg3 = (double)jarg3; 
  arg4 = (double)jarg4; 
  arg5 = (double)jarg5; 
  arg6 = (double)jarg6; 
  result = (EmbArcObject *)embArcObject_create(arg1,arg2,arg3,arg4,arg5,arg6);
  jresult = (void *)result; 
  return jresult;
}

// error in add dll
SWIGEXPORT void * SWIGSTDCALL CSharp_embPoint_make(double jarg1, double jarg2) {
  void * jresult ;
  double arg1 ;
  double arg2 ;
  EmbPoint result;
  
  arg1 = (double)jarg1; 
  arg2 = (double)jarg2; 
  result = embPoint_make(arg1,arg2);
  {
    EmbPoint * resultptr = (EmbPoint *) malloc(sizeof(EmbPoint));
    memmove(resultptr, &result, sizeof(EmbPoint));
    jresult = resultptr;
  }
  return jresult;
}

// error in add dll
SWIGEXPORT void * SWIGSTDCALL CSharp_embPointObject_make(double jarg1, double jarg2) {
  void * jresult ;
  double arg1 ;
  double arg2 ;
  EmbPointObject result;
  
  arg1 = (double)jarg1; 
  arg2 = (double)jarg2; 
  result = embPointObject_make(arg1,arg2);
  {
    EmbPointObject * resultptr = (EmbPointObject *) malloc(sizeof(EmbPointObject));
    memmove(resultptr, &result, sizeof(EmbPointObject));
    jresult = resultptr;
  }
  return jresult;
}

// error in add dll
SWIGEXPORT void * SWIGSTDCALL CSharp_embPointObject_create(double jarg1, double jarg2) {
  void * jresult ;
  double arg1 ;
  double arg2 ;
  EmbPointObject *result = 0 ;
  
  arg1 = (double)jarg1; 
  arg2 = (double)jarg2; 
  result = (EmbPointObject *)embPointObject_create(arg1,arg2);
  jresult = (void *)result; 
  return jresult;
}

//error in add dll
SWIGEXPORT void * SWIGSTDCALL CSharp_embPolygonObject_create(void * jarg1, void * jarg2, int jarg3) {
  void * jresult ;
  EmbArray *arg1 = (EmbArray *) 0 ;
  EmbColor arg2 ;
  int arg3 ;
  EmbColor *argp2 ;
  EmbPolygonObject *result = 0 ;
  
  arg1 = (EmbArray *)jarg1; 
  argp2 = (EmbColor *)jarg2; 
  if (!argp2) {
    SWIG_CSharpSetPendingExceptionArgument(SWIG_CSharpArgumentNullException, "Attempt to dereference null EmbColor", 0);
    return 0;
  }
  arg2 = *argp2; 
  arg3 = (int)jarg3; 
  result = (EmbPolygonObject *)embPolygonObject_create(arg1,arg2,arg3);
  jresult = (void *)result; 
  return jresult;
}

// error in 
SWIGEXPORT void SWIGSTDCALL CSharp_embPolygonObject_free(void * jarg1) {
  EmbPolygonObject *arg1 = (EmbPolygonObject *) 0 ;
  
  arg1 = (EmbPolygonObject *)jarg1; 
  embPolygonObject_free(arg1);
}

JoshVarga, 2021-09-11:
I sent you the 32 bit compiled SWIG for C# with the binding files

tatarize, 2021-11-03:
As a fun thing if you're quoting source code surround it with:
```
``C#
<source code>
``
```

Where C# is the language gets replaced with whatever language the source is in. Also, if you're an editor here or something you can go edit somebody else's post to correct this. So much easier to read.

---

That said, as far as the underlying reasons for this change I worked with fabrico on this topic but I think my resulting python code was a bit hard to translate.  See ( https://github.com/EmbroidePy/pyembroidery/blob/master/pyembroidery/EmbCompress.py ). This arises from the difficulties with the VIP and HUS formats using a defunct form of compression. Specifically it's ARJ compression with a 1024 window. This is the reason why most Embroidery Programs ship with a DLL called AL21DLL which is the compiled form of the code from Greenleaf, which was a technology company in the late 1990s and has since gone under, the remains were sold, and the founder died.

The only source code for how this compression worked was a closely guarded secret and they sold libraries of obfuscated source code to people so they could integrate this compression. Rather than including the AL21 compiled dll, libembroidery includes (the only copy of the obfuscated source code I could find for these functions. I found other scraps trying to chase down the source but I realized this is the copy-pasted source from basically orphaned software). However the ARJ compression format source does exist. And you can deobfuscate the source. But, ARJ is still a format run by the titular Robert Jung. And the legal issues surrounding this source code and my research on the topic lead me to recoding the decompression format from scratch in python. Since that's what pyembroidery is written in. See #124, for an example of how this can cause problems and how its almost impossible to fix without deep knowledge of what's going on.

Fabrico is trying to get HUS working in C# and the only code for how to decompress HUS files is written in Obfuscated C++ code that is impossible to translate. This was a really tough nut to crack and I did research and work on it on and off for about a year before I finally had a way forward. This involved tracking down the people who wrote the original code for Greenleaf and they pointed me to Jung, and reading the source code there and checking that the patents were expired and what they applied to. And writing my own code to do the same thing.

I even wrote my own compression algorithm that does 0% compression but writes a dummy table that gives you 1:1 table for compression. (2 bytes of chunk size), then (4 bytes: 0x02A001FE) will basically write a non-op table for compression from bytes to the exact same bytes. I could have written it 5 bits fewer but then the byte windows wouldn't line up (multiples of 8 is a lot better, so I also gave it a dummy lookback table to waste 5 bits).

robin-swift, 2021-11-05:
(I've applied the formatting as requested.)

Yes the current plan is to translate [your compression]( https://github.com/EmbroidePy/pyembroidery/blob/master/pyembroidery/EmbCompress.py ), I'll leave a message here when that's done.

robin-swift, 2021-12-01:
Ok, so I've added advice to the bindings section of the `README.md`.

closed 2021-12-01 by robin-swift