Embroidermodder/libembroidery #80: memory error protected

fabricocouto, 2021-09-14:
hello, the dll is being generated without errors using swig and visual studio 2019 c.
but when I make her call through csharp she gives memory error  in embpathern readprotected I've done many tests as admin and on another windows computer all with the same lock 

[image]

fabricocouto, 2021-09-14:
[image]

robin-swift, 2021-10-29:
This comes under shipping the new DLL (a general goal of 1.0), I'm closing as it's unclear that this will remain relevant.

closed 2021-10-29 by robin-swift