Embroidermodder/libembroidery #147: Command: Celtic Knots

robin-swift, 2021-11-04:
Copied from old to do list.

tatarize, 2021-11-04:
This idea properly belongs in Embroidermodder rather than libembroidery.

I did a series of investigations of this item in the todo list when I saw it, back when I was picking things out from this library for ideas.

While it's entirely actionable to make Celtic Knots in embroidery the sort of required code here is a grid where you turn segments on and off and the algorithm flows in a knot. But, to embroider that you'd actually need to lay down a series of satin stitches.

[image]

The results get you something like this, but you can't really do this as a single thread. Because what really matters there is the gaps, and you get these actually very fat sections with curves and flows and that's quite clearly going to be a satin stitch. While this idea is great it's *much* higher level than libembroidery.

http://www.mathrecreation.com/2018/11/algorithms-for-drawing-celtic-knots.html

fabricocouto, 2021-11-04:
I particularly find it very difficult to implement the dll in C# since it is among the 6 best languages today, however I can use a 2020 version of the dll, but even so I still use the codes written in C# which only lacks the file .hus and the .emb file

robin-swift, 2021-11-05:
@tatarize Thanks for linking that lovely article and js code by Dan MacKinnon, that will definitely be what I base the development on and I'll cite him in the documentation.

@fabricocouto We're talking about that in #69 and other threads, I can start a C# bindings issue but that's off-topic here.

robin-swift, 2025-07-31:
Moving to Embroidermodder because it'll fall under scripting there.