Embroidermodder/libembroidery #150: Command: Guilloche (Generates Guilloche patterns)

robin-swift, 2021-11-04:
Copied from old to do list.

tatarize, 2021-11-04:
This idea properly belongs in Embroidermodder.

From my investigations of this idea, the modern algorithms for Guilloche patterns absolutely does comport well with embroidery. But, in this case we're talking about things like step-repeat patterns to fill a curve between a cycloid with period of turn/5 and another with a period of turn/7 following a sine wave between the two boundary conditions.

http://www.excourse.com/excentro/ is one of the best programs for this, and shows the typical UI this involves. It requires a series of shapes that are blended together in various ways with particular effects. 

(Defunct Online Guilloche)
[image]

Basically you define a tree of different shapes descending from a base shape which is usually like a circle. Then you add to that circle an effect like a sine-wave, saw-tooth, or cycloid (dot rolling on a circle, rolling on the shape in question). You then calculate the position by figuring the location you would between 0,1 and shoe-horning that between the various other bases and effects, which form the envelope.

robin-swift, 2025-07-31:
Moving to Embroidermodder, along with other pattern generation algorithms.