Embroidermodder/libembroidery #180: Format Encoding and Needed Work/Changes for 1.0

tatarize, 2021-11-04:
I've written other embroidery input and output libraries for Python and Java. And considered this rather strongly over the years, ran into various problems with libembroidery itself and corrected them after considerable thought.

---

## Encoding

There is a need for a centralized set of methods to slightly abstract the embroidery between reading and writing (eg. https://github.com/EmbroidePy/pyembroidery/blob/master/pyembroidery/EmbEncoder.py). Without doing this you have the formats being hit with different commands and in different orders and some of these are not accepted by one format vs. another format. Sometimes the expectation is that you perform a stop, trim, then a series of jumps, other times the format may expect you to trim, jump, then stop for the color change. You can load from a format that has needle-set commands rather than color-change commands, and you need to know that the first needle-set there isn't a color change it's setting the first needle. Some formats permit stops commands others only use color changes. Sometimes you can fake a stop command such as with JEF. Some formats have explicit trim commands others, necessarily derive a trim command from 3+ jump commands.  If the format has explicit trims, you can't imply a trim because maybe that's a really long stitch for puffy fabric, and if it had an implied trim, that needs to be made into a *real* trim if your moving to a format with explicit trims. Some formats jump entirely to the place they are intending to stitch at then issue a stitch command. The stitch command means <needle, move> whereas some others the stitch command means <move, needle>. So you stitch then the dx,dy means move to this location and hit stitch rather than "after stitching" go to this location.

This is a lot of different bits nuance in code. And the only way I found to capture these nuances and be able to transfer formats from one to a fairly neutral format. This also means that you can add in some higher level commands without too much trouble. So if you want to do a TIE_OFF_THREE_STITCH to tie off a stitch block you can easily have those added, without messing with anything in the writers.

The writers in an encoded format would specify what type of formatting and ordering they expect and which commands they can accept. So if sequins are in the incoming data this can be stripped. Or if the data has speed commands these can be removed, and we can note that DSTs have a max jump length of 12.1mm and another might have no max jump length. This is a lot easier from a centralized location rather than trying to have every writer account for every nuance of everything.

There's also a lot of need to correct and work with the threadlist information. If you load a dst with 5 colors, but no color information, this gets dealt with a bit differently than colors that have actual discrete values of known information.

There's also some elements of metadata between the formats that is important. Sometimes label, author, copyright, etc. for particular formats can be written in other formats if they *also* have the same metadata. Though this usually doesn't require much encoding.

You'll notice that some items like #170 gets fixed pretty easily with this, because you'd no longer be feeding your raw stitches from one format into another format that might require radically different things. You say all the jumps and trims and whether there's a color change or not, gets called an end to the stitchblock or colorblock respectively and whatever the other format is can go ahead and have those elements converted right way for them. The writers can be guaranteed they will get consistent orderings and data types.

## Matrix

The code isn't hard but the effects are really powerful to have a solid set of matrix commands to apply affine transformations to code. E.g (https://github.com/EmbroidePy/pyembroidery/blob/master/pyembroidery/EmbMatrix.py). This gets really important when it comes to things like composing patterns together. #174 where you need to know where the next pattern should be relative to the previous pattern, or to perform basic changes on an embroidery like slightly resize it.

## Commands

There's a few more commands that occur within Embroidery than are accounted for within this library.  While higher level things like generic codes are needed some commands are actually seen in various embroidery formats and these need to be included to be effective.

* SEQUIN_MODE: This usually causes the sequin hopper to engage on the machine.
* SEQUIN_EJECT: This is often a modified jump, but accounting must be made for when they are triggered differently.
* STOP: Stop as opposed to color-change exists in some formats. The distinction is that stop causes the machine to stop but not necessarily change colors.
* SLOW: U01 formats lets you mark commands as slow. This causes the machine to run at a lower RPM until it it hits a FAST command which restores it to the same speed.
* FAST: U01 move back into higher RPM mode.

## NeedleSet

Some formats specifically .U01 (.UXY) and some others use a needle_set command rather than a color_change. This differs from color changes in 2 distinct and important ways. First, it sets the needle value, this occurs before sewing begins, rather than after sewing ends. And secondly it includes the needle number. This stuff needs to accounted for and permit the conversion between color-change (which is usually just STOP).

## STOP vs. COLOR_CHANGE

There are some formats that let you imply stopping without a color change. This is often for something like applying applique. We stop but we're not changing threads. Sometimes this even involves a FRAME_EJECT which is a trim and then moving to the top of the design to give the user unfettered access to the work to add the applique or whatever they are doing. 

## Versions

#139 is important. There needs to be given mechanisms for specifying versions of the file you're intending to write. Not just for things like PES v 6 but other methods of writing pes, like truncated. Or different versions of csv files. For example writing one that is actually correct to CSV formatting vs. one that maintains backwards compatibility, this gets to be important.

## Avoid Feature Creep.

A core library like this needs to just do the stuff it does in a well defined way. It doesn't need to do fill algorithms or anything that isn't dealing with the translations problems that can occur between different formats and those particular problems that can arise from them. If there is a need to have fills or something that should be spun off into a different more powerful library that does that work in a more generalized way since nothing about those fills would be format specific.

----

You will notice that *none* of that is that extreme. There's a lot of little problems with the readers and writers and little bugs and edge conditions. And those need to be addressed. But, really the biggest issues there are a couple of small things and the ability to sort of generalize the file between reading and writing. Additional features like composing two different files together get a lot easier, when this is done correctly. And the bulk of the important work is understanding the formats themselves, which is mostly already done.

This is one part of the program that can just be finished, with tiny format bug fixes that are very isolated to just the reader of such and such a format. But, that requires that we get rid of things like all writers need to check if they have an end command, or all writers need to deal with sequin stitches explicitly or this reader read this stuff verbatim but that's the wrong order for this writer.

tatarize, 2021-11-05:
Suffice it to say, I don't believe any of the fill or the satin stuff, etc stuff belongs here. This should really focus on the Embroidery reading and writing, in a way that isn't infinitely complex or subject to feature creep.

You can really zero in on the main very low level reading and writing stuff make it rock solid.

If there's a need to have the fill algorithm stuff divided from the gui stuff it might be best to put that in a different library that could simply depend on this one, though it would be a library of fill-like commands to points which doesn't even need to know what embroidery formats are.

robin-swift, 2021-11-05:
There's a lot here that is useful so I feel it's better placed in the documentation. Can you make a pull request on the Encoding header to the end of the Avoid Feature Creep section as part of the `docs/` folder? I'll need to refer to details like this and the printer-friendly docs are going to be my first point of reference. You can just make a new file if you don't want to get into editing the TeX.

Also I find large blocks of san-serif, variable width fonts particularly hard to read (I lose my place a lot, probably mild visual dyslexia) so a printer-friendly version of your longest comments where I can change the formatting would be very helpful to me.

tatarize, 2021-11-05:
I'm not sure it belongs in documentation. But, I went over it and removed some of the unneeded words and phrases and broke up the text into more relevant shorter sections with more clearly demarcated grammar.