Embroidermodder/libembroidery #190: PES to SEW conversion not working

BartmanEH, 2022-01-25:
fresh build of libembroiderery on mac 12.1 but conversion from PES to SEW produces invalid results

robin-swift, 2022-01-25:
Ok, can you show me the terminal output from the conversion?

Let's work this out.

BartmanEH, 2022-01-25:
```
 ./libembroidery/embroider test.pes test.sew                                                       
test.pes
```
maybe I'm missing `-t sew`?

robin-swift, 2022-01-25:
No that should work. What's the file size of the output?

BartmanEH, 2022-01-25:
22KB

robin-swift, 2022-01-25:
Does the sew file not load at all? Or is it incomplete?

BartmanEH, 2022-01-25:
using StitchBuddy, the PES previews fine but the SEW is blank

robin-swift, 2022-01-25:
Ah, that's the error message I need. Can you paste the output of:

```
% xxd test.sew | head
```

I.e. the first 256bytes in hex format.

BartmanEH, 2022-01-25:
```
00000000: 6004 0000 0a00 0000 1d00 0000 2900 0000  `...........)...
00000010: 0100 0000 3a00 0000 1d00 0000 2900 0000  ....:.......)...
00000020: 0100 0000 4200 0000 1d00 0000 1e00 0000  ....B...........
00000030: 0100 0000 0c00 0000 1d00 0000 1e00 0000  ................
00000040: 0100 0000 4200 0000 1d00 0000 1e00 0000  ....B...........
00000050: 0100 0000 0100 0000 0100 0000 3200 0000  ............2...
00000060: 0100 0000 0100 0000 0100 0000 3200 0000  ............2...
00000070: 0100 0000 3200 0000 1d00 0000 3200 0000  ....2.......2...
00000080: 0100 0000 3200 0000 1d00 0000 3200 0000  ....2.......2...
00000090: 0100 0000 3200 0000 1d00 0000 3200 0000  ....2.......2...
```

BartmanEH, 2022-01-25:
I tried DST conversion but it doesn't work either and I get a lot of screen output:
```
 libembroidery % ./embroider test.pes test.dst
test.pes
0 90.933333 19.111111 909 191 909 191
ERROR: format-t01.c encode_record(), x is not in valid range [-121,121] , x = 909
1 79.566667 16.722222 -113 -24 796 167
2 68.200000 14.333333 -114 -24 682 143
3 56.833333 11.944444 -114 -24 568 119
4 45.466667 9.555556 -113 -23 455 96
5 34.100000 7.166667 -114 -24 341 72
6 22.733333 4.777778 -114 -24 227 48
7 11.366667 2.388889 -113 -24 114 24
8 102.300000 21.500000 909 191 1023 215
ERROR: format-t01.c encode_record(), x is not in valid range [-121,121] , x = 909
```
[truncated]

robin-swift, 2022-01-25:
Yeah, that's big jumps not being split properly. I know about that bug.

If you hang on 5 minutes I may work out what this colour change bug is.

robin-swift, 2022-01-25:
Can you paste the output of

```
% ./libembroidery/embroider test.pes test.csv
% head test.csv
```

BartmanEH, 2022-01-25:
I'm not a dev, I'm a wannabe so that last command didn't work but this did:
```
% head csv
head: csv: No such file or directory
bastianstassen@iMac-659 ~ % xxd test.csv | head
00000000: 2223 222c 2245 6d62 726f 6964 6572 6d6f  "#","Embroidermo
00000010: 6464 6572 2032 2043 5356 2045 6d62 726f  dder 2 CSV Embro
00000020: 6964 6572 7920 4669 6c65 220a 2223 222c  idery File"."#",
00000030: 2268 7474 703a 2f2f 656d 6272 6f69 6465  "http://embroide
00000040: 726d 6f64 6465 722e 6769 7468 7562 2e69  rmodder.github.i
00000050: 6f22 0a0a 2223 222c 2247 656e 6572 616c  o".."#","General
00000060: 204e 6f74 6573 3a22 0a22 2322 2c22 5468   Notes:"."#","Th
00000070: 6973 2066 696c 6520 6361 6e20 6265 2072  is file can be r
00000080: 6561 6420 6279 2045 7863 656c 206f 7220  ead by Excel or 
00000090: 4c69 6272 654f 6666 6963 6520 6173 2043  LibreOffice as C
```

robin-swift, 2022-01-25:
So what I want to see is slightly below that:

```
% head -n40 test.csv
```

robin-swift, 2022-01-25:
Oh dear, that's a new one. I've not cleared out the memory for the strings so it's got whatever rubbish is there.

I'm writing a patch that should help with this testing and fix some minor bugs, hang on.

robin-swift, 2022-01-25:
Ok, `git pull` and rebuild with `cmake --build .`.

BartmanEH, 2022-01-25:
oh man... I'm not a dev, I'm a hack... this is gonna go horribly wrong :-)

robin-swift, 2022-01-25:
You've followed me to here I've not been confused by your replies at any point.

This is a good real world test of the software, so I'm enjoying it.

robin-swift, 2022-01-25:
You do `git pull` from inside the code directory.

So from where your terminal appears to be:

```
cd libembroidery
git pull
```

BartmanEH, 2022-01-25:
```
% ./libembroidery/embroider test.pes test.sew
test.pes
Debugging Information
number of colors = 251
number of stitches = 3896
Color: 0 1
Color: 1 1
Color: 2 1
Color: 3 1
Color: 4 1
Color: 5 1
Color: 6 1
```
[truncated]

robin-swift, 2022-01-25:
Huh, now that's surprising.

How similar to your design is

```
number of colors = 251
number of stitches = 3896
```

I doubt it really is a 251 thread color pattern.

BartmanEH, 2022-01-25:
lol, indeed not: 2 colors

robin-swift, 2022-01-25:
and how many stitches?

BartmanEH, 2022-01-25:
[image]

robin-swift, 2022-01-25:
That's useful. I'm thinking...

robin-swift, 2022-01-25:
Ok, so can I see the same hexdump for your input?

```
xxd test.pes | head
```

BartmanEH, 2022-01-25:
```
% xxd test.pes | head
00000000: 2350 4553 3030 3031 802f 0100 0100 0100  #PES0001./......
00000010: 0100 ffff 0000 0700 4345 6d62 4f6e 6500  ........CEmbOne.
00000020: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000030: 0080 3f00 0000 0000 0000 0000 0080 3f00  ..?...........?.
00000040: d80e 4480 eba5 4401 0000 0000 0059 038e  ..D...D......Y..
00000050: 0200 0000 0000 0000 0062 00ff ff00 0007  .........b......
00000060: 0043 5365 7753 6567 0100 1d00 0200 ac01  .CSewSeg........
00000070: b9fe 0400 51ff 0380 0000 1d00 872c 0400  ....Q........,..
00000080: 51ff 0000 52ff 0400 51ff 0000 52ff 0400  Q...R...Q...R...
00000090: 51ff 0000 52ff 0a00 5cff 1500 67ff 2300  Q...R...\...g.#.
```

robin-swift, 2022-01-25:
Ok, so I think I can offer your workflow a solution quicker than fixing this. Then fix this at a later date.

Can you output from your design software in Tajima `.dst`? Then do the conversion to `.sew`.

BartmanEH, 2022-01-25:
OK, I'll try that but it will take me some time to wrestle Inkstitch into submission, again. Thanks for your support. I'll post my results here hopefully later tonight.

robin-swift, 2022-01-25:
I'll open an new issue... use #191

JoshVarga, 2022-01-25:
Based on the output you provided, SEW is really not outputting correctly. those first two bytes couldn't possibly be that large. I made a change to output the header with the correct values. That should get this MUCH closer to working... Maybe give it another shot.
> ```
> 00000000: 6004 0000 0a00 0000 1d00 0000 2900 0000  `...........)...
> 00000010: 0100 0000 3a00 0000 1d00 0000 2900 0000  ....:.......)...
> 00000020: 0100 0000 4200 0000 1d00 0000 1e00 0000 [...]

BartmanEH, 2022-01-25:
after a git pull and make:
```
% ./libembroidery/embroider test.pes test.sew
test.pes
Debugging Information
number of colors = 251
number of stitches = 3896
```
output is 21KB now but still not viable.