Embroidermodder/libembroidery #191: Confirm DST to SEW conversion

robin-swift, 2022-01-25:


BartmanEH, 2022-01-25:
```
% ./libembroidery/embroider test.dst test.sew
Failed to open file with name: test.edr.Failed to open file with name: test.rgb.Failed to open file with name: test.col.Failed to open file with name: test.inf.Debugging Information
number of colors = 1
number of stitches = 19190
Color: 0 1
```

JoshVarga, 2022-01-25:
I fixed a big issue with SEW write, but that doesn't address the .DST read. We should not be requiring color files to exist. If no color files exist we should just use random colors or a fixed palette.

robin-swift, 2022-01-25:
That's debug information, not the reason for failing.

It looks like the dst parsing is getting the number of stitches correct?

Does the sew file open now?

BartmanEH, 2022-01-25:
> Does the sew file open now?

nope

BartmanEH, 2022-01-25:
but wait! when I try with the DST I get closer now:
[image]
it opens in StitchBuddy but it's just one color instead of the 2 expected. Progress!

robin-swift, 2022-01-26:
Yes, I can reproduce this bug...

robin-swift, 2022-01-26:
Ok, try the current version. In my testing it managed to encode multiple colors.

BartmanEH, 2022-01-26:
```
% ./libembroidery/embroider test.dst test.sew
ERROR: Failed to open file with name: test.edr.
ERROR: Failed to open file with name: test.rgb.
ERROR: Failed to open file with name: test.col.
ERROR: Failed to open file with name: test.inf.
Debugging Information
number of colors = 1
number of stitches = 19190
```
SEW file is 84KB and not working at all

BartmanEH, 2022-01-26:
```
% ./libembroidery/embroider test.pes test.sew
debug information for reading fileName: test.pes
pecstart = 12160
Debugging Information
number of colors = 251
number of stitches = 3896
```
SEW file is 23KB and not working at all

robin-swift, 2022-01-26:
Oh dear, that's not it then. What does

```
% ./libembroidery/embroider test.dst test.csv
% head -n40 test.csv
```

give now?

BartmanEH, 2022-01-26:
```
% head -n40 test.csv
"#","Embroidermodder 2 CSV Embroidery File"
"#","http://embroidermodder.github.io"

"#","General Notes:"
"#","This file can be read by Excel or LibreOffice as CSV (Comma Separated Value) or with a text editor."
"#","Lines beginning with # are comments."
"#","Lines beginning with > are variables: [VAR_NAME], [VAR_VALUE]"
"#","Lines beginning with $ are threads: [THREAD_NUMBER], [RED], [GREEN], [BLUE], [DESCRIPTION], [CATALOG_NUMBER]"
"#","Lines beginning with * are stitch entries: [STITCH_TYPE], [X], [Y]"

"#","Stitch Entry Notes:"
"#","STITCH instructs the machine to move to the position [X][Y] and then make a stitch."
"#","JUMP instructs the machine to move to the position [X][Y] without making a stitch."
"#","TRIM instructs the machine to cut the thread before moving to the position [X][Y] without making a stitch."
"#","COLOR instructs the machine to stop temporarily so that the user can change to a different color thread before resuming."
"#","END instructs the machine that the design is completed and there are no further instructions."
"#","UNKNOWN encompasses instructions that may not be supported currently."
"#","[X] and [Y] are absolute coordinates in millimeters (mm)."

"#","[VAR_NAME]","[VAR_VALUE]"
">","STITCH_COUNT:","19190"
">","THREAD_COUNT:","1"
">","EXTENTS_LEFT:","-42.900000"
">","EXTENTS_TOP:","-32.700000"
">","EXTENTS_RIGHT:","42.900000"
">","EXTENTS_BOTTOM:","32.700000"
">","EXTENTS_WIDTH:","85.800000"
">","EXTENTS_HEIGHT:","65.400000"

"#","[THREAD_NUMBER]","[RED]","[GREEN]","[BLUE]","[DESCRIPTION]","[CATALOG_NUMBER]"
"$","1","0","0","0","Black","000"

"#","[STITCH_TYPE]","[X]","[Y]"
"*","JUMP","0.000000","0.000000"
"*","JUMP","-10.600000","-3.800000"
"*","JUMP","-21.200000","-7.600000"
"*","JUMP","-31.800000","-11.400000"
"*","JUMP","-42.400000","-15.200000"
"*","STITCH","-42.400000","-15.200000"
"*","STITCH","-42.900000","-15.300000"
```

robin-swift, 2022-01-26:
So the question is, what happens when the dst attempts to encode a color change? Is it an unknown?

What does this return?

```
% grep UNKNOWN test.csv 
```

BartmanEH, 2022-01-26:
```
% grep UNKNOWN test.csv 
"#","UNKNOWN encompasses instructions that may not be supported currently."
"*","UNKNOWN","-17.700000","-25.200000"
```

robin-swift, 2022-01-26:
That looks like a color change to me. There's one of them for a start.

Perhaps an UNKNOWN should be a STOP on read? Because if we don't know that the user wants to do the machine should stop operation.

I've made that change. Try pulling the current version then:

```
% ./libembroidery/embroider test.dst test.csv
% ./libembroidery/embroider test.csv test.sew
```

The color information will break, but when you run it on your machine it should not matter. I think this works as a hack for now.

BartmanEH, 2022-01-26:
```
 % ./libembroidery/embroider test.dst test.csv
ERROR: Failed to open file with name: test.edr.
ERROR: Failed to open file with name: test.rgb.
ERROR: Failed to open file with name: test.col.
ERROR: Failed to open file with name: test.inf.
bastianstassen@iMac-659 ~ % ./libembroidery/embroider test.csv test.sew
Debugging Information
number of colors = 2
number of stitches = 19191
```
it looks better debug-info-wise, but the SEW file is 84KB and not viable

robin-swift, 2022-01-26:
Ok, after some more testing I've routed out more bugs. Can you try it?

BartmanEH, 2022-01-26:
```
~% ./libembroidery/embroider test.dst test.csv
ERROR: Failed to open file with name: test.edr.
ERROR: Failed to open file with name: test.rgb.
ERROR: Failed to open file with name: test.col.
ERROR: Failed to open file with name: test.inf.
~% ./libembroidery/embroider test.csv test.sew
zsh: abort      ./libembroidery/embroider test.csv test.sew
~% ./libembroidery/embroider test.csv test.sew
zsh: abort      ./libembroidery/embroider test.csv test.sew
~% 
```

BartmanEH, 2022-01-27:
just checkin' in with my embroidery file conversion expert crew....

robin-swift, 2022-01-27:
So I've had a few attempts at this. I'm playing whack-a-mole with bugs and eventually one of them will be the one that's causing this problem.

Generally, my development has taken this source code from 24,000 lines to around 16,000 while adding features and in the process it made these bugs and underdeveloped features easy to fix.

This isn't a profit making enterprise, so we've started taking donations to help us spend more time on this (I'm an artist, I don't code for wages). People who donate to us will get emails about what has changed over the last month and we will prioritise what the backers want. If you donated I would help keep this a priority as the SEW format as this isn't one I've done much on and it's a relatively rare format to use.

So yeah, sorry this wasn't a simple fix, but we will try to fix this eventually even without donations. We want this software to be good and people to have more creative control over their machines and designs.

BartmanEH, 2022-01-27:
OK, understood. That puts me in a bit of an awkward spot.

On the one hand, I do believe your work is worth something. On the other hand, I currently have access to a friends old Kenmore 19000 that only supports SEW format. Ultimately I will move my projects to a public Makerspace with a modern Brother machine (currently closed due to public health measures). So my needs for SEW format are also rather short term and transient. Also, I can get by with a freeware combination of Inkscape/Inkstitch on macOS and Bernina ArtLink in a Windows VM.

There are many commercial options available and even StitchBuddy is "only" $70 and this would provide the ability to convert to SEW. Unfortunately this hobby is far down the list of my hobbies that I care to invest in. I came across this project that looked rather well established and thought it would work out of the box. It appeared to have been around for many years. I'm not really sure what the project's aims are but I gave it a shot and it seems to not work for SEW format unfortunately.

I hope I haven't dragged you too far down a rabbit hole of little value to your project's ultimate endeavours.

robin-swift, 2022-01-27:
Oh no, having people try out specific real world tasks is good for finding out what needs doing and there have been improvements already.

As for "well established" it's complicated: development hasn't been active the whole time. I joined about 6 months ago. There is a lot of work behind where we're at and the task given only hobbyists hours is quite large for good reason.

Check in in a year or so and maybe you'll want to use the embroider CLI and Embroidermodder 2 in your toolkit by then.