Embroidermodder/libembroidery #199: Support Wilcom's .emb native format

robin-swift, 2022-04-03:


robin-swift, 2022-04-03:
Seems we already have some binaries to pull apart in the `samples/` folder (e.g. [this one](https://github.com/Embroidermodder/Embroidermodder/blob/main/samples/star/Star.EMB)). I'll post some interpretation here if I get anywhere.

tatarize, 2022-04-03:
https://stackoverflow.com/questions/3488544/embroidery-file-formats

Has some of my edited results over the last few years. It might be possible to read most of the .emb file. It's Compound Binary File Format, with an `Contents` file that you can read as it's 4 bytes of size and then a zlib stream. This gives you a bunch of uncompressed proper data and then it appears to be a triplet system within that. 

The more important stuff the Design stuff which is apparently encrypted since I can't find a stream in it but the values of the binary are equally distributed.

robin-swift, 2022-04-03:
Huh, that sounds challenging, I'll have to tag this as "version 1.1 or later".

Encrypted files without any official file format description should generally be done after we get the Python bindings working and simpler file formats working to some level.

JohnMPC, 2022-04-19:
Hi Guys, Thanks for taking a look at this. I have PLENTY of .EMB files that i can provide if you need. 
The idea behind what i am working on came from https://embtrak.com/. If you haven't already seen this, could be worth a look for some ideas.

robin-swift, 2022-04-24:
@JohnMPC The best way to add a new format is to have a complex example of that format along with a rendering of what that format looks like preferably showing the individual stitches. Here's a little primer so I can paste the advice into the docs.

> ### How to Help Develop A Specific Format
> 
> For example, if I wanted to test/add a feature to our PES parser I'd open up another program that supports PES, create a non-trivial example like the ones in our `Embroidermodder/samples/` folder then export 3 copies:
> 
> 1. The .pes file `example.pes`
> 2. A simpler format version so we can check that our interpretation matches: `example.csv`. (Chosen because it's both machine and human readable.)
> 3. A render in a common image format `example-render.bmp`.
> 
> If you could do this for any your files you're comfortable sharing for this purpose, preferably about 5 examples showing different features (like automated fills or other weird features that may be supported), then that would help us improve your format.
> 

We're not starting from scratch because @tatarize has compiled notes above and [here](https://stackoverflow.com/questions/3488544/embroidery-file-formats) but I'd definitely do more to make this happen with these examples to hand.

fabricocouto, 2024-11-04:
Wilcom itself wrote a dll capable of reading its native emb files in trueview.
I was able to read the emb files using their own dll,

[image]

fabricocouto, 2024-11-12:
> Oi pessoal, Obrigado por dar uma olhada nisso. Eu tenho MUITO . Arquivos EMB que eu posso fornecer se você precisar. A ideia por trás do que estou trabalhando veio de https://embtrak.com/. Se você ainda não viu isso, pode valer a pena dar uma olhada em algumas ideias.

In fact, nothing guarantees that they have encrypted the EMB file, or natively open the EMB file in their software, they can also use  wilcom api.
I even managed to read the EMB file up to version 4.2 with their own dll without even having wilcom installed just using the dll they already have in wilcom

putnam, 2026-09-18:
> In fact, nothing guarantees that they have encrypted the EMB file, or natively open the EMB file in their software, they can also use wilcom api. I even managed to read the EMB file up to version [...]

What DLL would that be? Do you recall the version?