Embroidermodder/libembroidery #202: Support for wings ngs embroidery format

andmaltes, 2022-05-18:
Hi, Are you planning to add support to wings ngs embroidery format? if it is not in the plan, I would be up to contribute on this with some guidance, thanks!

TruGolfJV, 2022-05-18:
I have looked at NGS in the past I don't think I was able to make much progress other than reading some of the metadata. IIRC it uses [MS-CFB: Compound File Binary File Format](https://docs.microsoft.com/en-us/openspecs/windows_protocols/ms-cfb/53989ce4-7b05-4f8d-829b-d08d6148375b) I am not certain but I believe the files inside that are potentially encrypted. It is worth another look, but we haven't had much demand for it. Are you a software dev?

JoshVarga, 2022-05-18:
I have looked at NGS in the past I don't think I was able to make much progress other than reading some of the metadata. IIRC it uses [MS-CFB: Compound File Binary File Format](https://docs.microsoft.com/en-us/openspecs/windows_protocols/ms-cfb/53989ce4-7b05-4f8d-829b-d08d6148375b) I am not certain but I believe the files inside that are potentially encrypted. It is worth another look, but we haven't had much demand for it. Are you a software dev?

andmaltes, 2022-06-01:
Yes, I am a software dev. But would be my first time working with files formats and reverse engineering