Embroidermodder/libembroidery #207: Various issues on MacOS

e-e-e, 2022-10-04:
I am new to machine embroidery, but have been looking for good libraries to help programatically generate patterns.
This library looks wonderful - so I cloned it and attempted to play with it locally.
I am a mac user and the cli tool compiles. However when running it there are a lot of issues that it's difficult for me to isolate and diagnose.

1. `-to` flag fails with `Please enter an output format for your file, see --help.` this is using the command `embroider -t dst cmmn.jef `

2. `--test` and `--full-test-suite` fails with `[1]    61997 segmentation fault  ./embroider --test`

3. `--render` fails with:
    ```
    228 
    [1]    62063 segmentation fault  ./embroider -r cmmn.jef test.png
    ```
4. Basic conversion passes without seg faults -  using the command `emboider cmmn.jef cmmn.txt`. However, converting to csv is showing indications of where segfault may be coming from as the "description"  field for threads are showing some weirdness:
     ```
     "#","[THREAD_NUMBER]","[RED]","[GREEN]","[BLUE]","[DESCRIPTION]","[CATALOG_NUMBER]"
     "$","1","115","58","0","    -c, --circle     Add a circle defined by the arguments given to the current  to the current ��","rguments given to the current  to the current ��"
     "$","2","0","0","0","Black","Black"
     ```
   and for another file it gives:
   ```
    "#","[THREAD_NUMBER]","[RED]","[GREEN]","[BLUE]","[DESCRIPTION]","[CATALOG_NUMBER]"
    "$","1","101","100","32","under the terms of the zlib license.
    
        https://github.com/Embroidermodder/lib","github.com/Embroidermodder/lib"
    "$","2","45","100","105","fat.c bcf_difat_create(), Unexpected sector value %x at DIFAT[%d] ","%x at DIFAT[%d]"
    "$","3","0","0","0","END","END"
   ```

5. In good news `embroider -F` and `embroider --help` is working as expected.

My basic hardware settings are:
```
  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro16,1
  Processor Name:	8-Core Intel Core i9
  Processor Speed:	2.4 GHz
  Number of Processors:	1
  Total Number of Cores:	8
  L2 Cache (per Core):	256 KB
  L3 Cache:	16 MB
  Hyper-Threading Technology:	Enabled
  Memory:	32 GB
 ```
 Running OSX 12.6
 
Let me know if there is any other information that would be helpful to provide?

I am interested in your project and would love to contribute anyway I can. Not a solid C developer.

robin-swift, 2022-10-05:
Hi @e-e-e 

It's good to hear from someone interested in our project, especially under the hood. How did you discover the project? I've have a bunch of messages in the last 4 days and I don't know why. I assume it's some content aggregator.

The library is very much in development (version 1.0 has no planned release date since it's pretty much just me writing it at the moment and we don't get a lot in donations), so there will be bizarre segfaults and stack overflows. The best way to report them is as separate issues with backtrace. For example the valgrind output from:

```
$ valgrind ./embroider --test
```

along with the description of your system. If you've compiled with the debug symbols (this is the default) then it should give you specific line numbers. Another option is the GNU debugger which is

```
$ gdb --args ./embroider --test
```

which drops you into a new shell where you can inspect the runtime during execution. The commands `r` and `bt` will run and backtrace your program respectively.

So your report here would be more like 4 reports. Which would be a back and forth as we try to fix them.

Help with documentation is also useful at the moment, but that is housed on our other repository https://github.com/Embroidermodder/embroidermodder in the docs folder.

Is there anything in particular that you'd really like to see this library do?

Cheers,

Robin

e-e-e, 2022-10-05:
Thanks for getting back so quickly @robin-swift. I did have a quick look at the latest commits and figured you were early stages, and still very much WIP. I hoped sharing the issues I found might be helpful.

I ended up at this project through looking at embroidery formats - thinking I would have to hack something myself, then I found inksketch and the lovely work on pyembroidery by @tatarize. Their project cites the origins as coming from this work via the java implementation.

I have been toying with the idea of building a little ipad app to help me generated embroidery from drawings, but obviously python and ios is not a natural fit, so I started to dig into your project to see where it was at. Did not want to reinvent the wheel, and thought that I may be able to help contribute to another projects development.

> Is there anything in particular that you'd really like to see this library do?

Ideally all I have been looking for is a simple way to translate low level commands into common embroidery formats.
The preview and simulation features are useful.

As for the bugs above - I will see if I can get stack trace information using the instructions you have provided and open separate issues here. Once I open them I will close this issue. 

Thanks so much for open sourcing your work. I really appreciate the labour in support of the commons.

robin-swift, 2022-10-06:
Oh man, I left this real broken. So the original intension was to share a lot of code between the embedded (i.e. embroidery machine) code, the C library and the CLI. So what I had done was separated out a bunch of data into CSV files to reduce the compiled binary size but haven't finished setting up how to access that data.

My latest commit fixes the to flag parsing which was just missing. I'll have a look at those segfaults.

robin-swift, 2022-10-06:
> As for the bugs above - I will see if I can get stack trace information using the instructions you have provided and open separate issues here. Once I open them I will close this issue.

The last commit should fix some more problems: be sure to `git pull` before you do more testing.

robin-swift, 2025-04-03:
The new approach is in #237 this version is out of date.

@e-e-e is welcome to join in this new version which is targeted at reducing compiler warnings. Passing testing is a task for the future.

closed 2025-04-03 by robin-swift