Embroidermodder/libembroidery #230: Function to return an array with all points and RGB colors.

fabricocouto, 2024-11-09:
// Function to return an array with all points and RGB colors.
EMB_PUBLIC EmbStitchInfo* emb_pattern_getAllStitches(EmbPattern* pattern, int* stitchCount) {
    if (!pattern || !pattern->stitch_list || !stitchCount) return NULL;

    *stitchCount = pattern->stitch_list->count; // Define o número de pontos
    EmbStitchInfo* stitches = (EmbStitchInfo*)malloc(*stitchCount * sizeof(EmbStitchInfo));

    if (!stitches) return NULL;  

   
    for (int i = 0; i < *stitchCount; i++) {
        stitches[i].x = pattern->stitch_list->stitch[i].x;
        stitches[i].y = pattern->stitch_list->stitch[i].y;
        stitches[i].flags = pattern->stitch_list->stitch[i].flags;

 
        int colorIndex = pattern->stitch_list->stitch[i].color;
        if (colorIndex < pattern->thread_list->count) {
            EmbColor color = pattern->thread_list->thread[colorIndex].color;
            stitches[i].red = color.r;
            stitches[i].green = color.g;
            stitches[i].blue = color.b;
        }
        else {
            // Cor padrão caso o índice não seja válido
            stitches[i].red = 0;
            stitches[i].green = 0;
            stitches[i].blue = 0;
        }
    }

    return stitches;  
}

EMB_PUBLIC void emb_freeStitchInfoList(EmbStitchInfo* stitches) {
    free(stitches);
}

robin-swift, 2024-11-19:
I don't quite understand what this is for. Can you give me a simple use case, please?

fabricocouto, 2024-11-19:
> I don't quite understand what this is for. Can you give me a simple use case, please?

[image]
So that I could view the colors correctly in C#, I don't know why but I had to do that

var i = 0;
 for (i = 0; i < stitchCount; i++)
 {
     IntPtr stitchPtr = IntPtr.Add(stitchesPtr, i * Marshal.SizeOf<EmbStitchInfo>());
     stitches[i] = Marshal.PtrToStructure<EmbStitchInfo>(stitchPtr);
     var flags = (StitchType)stitches[i].flags;
     if (flags == StitchType.fSTOP || flags == StitchType.colorchnge)
     {
         var colors = Color.FromArgb(255,stitches[i-1].red, stitches[i-1].green, stitches[i-1].blue);
         patternclass. AddColor(ref colors);
     }
  }

I already have a separate viewer in C# so I would only need the point listing and the color listing.

there is a reading error because the format does not read correctly, I haven't stopped yet to check the code