Embroidermodder/libembroidery #232: cmake --build . gives error: "implicit declaration of function 'emb_fwrite'"

sercan5534, 2024-12-28:
I'm not a C/C++ guy, but I followed steps on Readme:
- first checkout the repo with current branch
- than, I execute "cmake ."
- When I try to execute "cmake --build .", it gives me:                                                           ^
/Users/sercanozen/libembroidery/src/pattern.c:906:5: error: implicit declaration of function 'emb_fwrite' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    emb_fwrite(&flagsToWrite, 1, file);
    ^
I'm using Mac M1(silicon) and I searched "emb_fwrite" function and didn't found anywhere.

jeremiahgivens, 2025-03-22:
I am facing the same issue on this and EmbroiderModder2, and I am on the same hardware.

robin-swift, 2025-04-01:
This is fixed in `libembroidery` now: the function no longer exists, it isn't called anywhere. I switched `embroidermodder` back to using a `libembroidery` submodule so it should be fixed there too.

closed 2025-04-01 by robin-swift